riffnshred/nhl-led-scoreboard #280: Board erroring out due to playoffs

Coop21, 2021-05-13:
**Describe the bug**
Board is not starting up as of today.  Note: I changed my default from a non-playoff team (Kings) to a playoff team (Blues) and the board would still not work.  See screen shot of error below.  

**Logs**
[image]

pjockey, 2021-05-14:
what does your config.json look like? Can you post?

riffnshred, 2021-05-14:
This is an issue caused by some playoff data missing in the API. I got a fix for it, but other issues came out with it and I need to wait and see how the playoff data will settle till this weekend before I do an official release. I will have it in the Beta branch soon tho.

riffnshred, 2021-05-14:
Fix is on the Beta

justinaramian, 2021-05-14:
I woke up this morning to find my board was now working again. Didn't make any changes. Just let it sit in a boot loop over night.

riffnshred, 2021-05-20:
All seems normal now. closing.

cheers

closed 2021-05-20 by riffnshred