riffnshred/nhl-led-scoreboard #284: 1.6.4. update breaks board rendering

rentiak, 2021-05-16:
**Describe the bug**
The latest playoff 1.6.4 release breaks rendering of certain text fields.

[image]

[image]

riffnshred, 2021-05-16:
This is caused by the latest update to Pillow , which is what we use to generate the image that the screen render.

try uninstalling it 

sudo pip3 uninstall pillow

and then rerun the update procedure.

```
git reset --hard
git checkout master
git pull
chmod +x scripts/install.sh
./scripts/install.sh
```

rentiak, 2021-05-16:
That worked for me. Series ticket and weather board (pictures above) both render correctly now.

riffnshred, 2021-05-16:
Perfect. Thanks bud !

closed 2021-05-16 by riffnshred