riffnshred/nhl-led-scoreboard #289: Scoreticker cycling slow

jeepdrvr71, 2021-05-20:
**Describe the bug**
I set the scoreticker cycle rate to 1 to speed up the cycling and it doesn't seem to make a difference. Is 1 equal to 1 minute?

 

**Hardware you are using**
- Pi Zero
-  Adafruit HAT
- Anti-flicker = Yes

**Software**
1.6.5

riffnshred, 2021-05-20:
the Pi Zero is the issue. What you see is Lag because it only has one Core instead of 4 like the Pi 3 and up. There is nothing you can do to fix this other than upgrading.

jeepdrvr71, 2021-05-20:
Good to know. If I switch to a Pi 3can I use the same hat as the Pi Zero?

riffnshred, 2021-05-20:
Definitely. I recommend reading the Hardware section in the wiki, there are some important details regarding power usage. There are some things to look for and things I recommend to not do so you keep away from any potential issues.

ThatOneRoadie, 2021-05-21:
It might be worth overclocking; I'm still getting accurate cycle times on a Pi Zero W using this overclock, the adafruit hat, and no heatsink (but my Pi is mounted _vertically_ on the back of the matrix, which may have a non-negligible impact on heat dissipation even without a heatsink):

Add to the bottom of /boot/config.txt and restart:
```
#overclock
arm_freq=1085
#arm_freq=1095 for my other unit with heatsink
gpu_freq=530
#gpu_freq=550 for the unit with a heatsink.
over_voltage=2
core_freq=515
sdram_freq=533
over_voltage_sdram=1
```