riffnshred/nhl-led-scoreboard #297: missing Partition table

may2700, 2021-06-14:
**Describe the bug**
im FINALLY trying to get mu board up and running again by using the image method. when i go into etcher and select the img file I get an error that says "It looks like this is not a bootable image. the image does not appear to contain a partition table, and might not be recognized or bootable by your device"

**Logs**
no logs since i cant  get anything running

**Your config file**
again, cant even get the i age on the card

**Hardware you are using**
- Raspberry pi model - 3B
- The Matrix driver - Adafruit bonnet
- Anti-flicker mod or not? - yes

**Software**
- Which version of the software are you on?
trying to install 1.6.5

may2700, 2021-06-14:
not that this is a solution, but i did the manual install and got the board up and running. still interested why i was getting this error.

falkyre, 2021-06-18:
Try using the official raspberry pi imager. It's been fully tested and works with that for sure.