riffnshred/nhl-led-scoreboard #304: Board Issue for Pi 3+

kempdawg20, 2021-08-28:
**Describe the bug**
Not getting a clear image of the board. The board and adafruit hat were working fine on my Pi Zero. I did a brand new software install on the SD card, got a spacer to make sure that the hat wasn't touching anything else on the Pi. I tried to launch with a different couple of commands. The first one was sudo python3 src/main.py --led-gpio-mapping=adafruit-hat --led-brightness=50 --led-slowdown-gpio=2 --led-parallel=1 which only rendered a couple of lines that changed. I then changed the slowdown to 4 which now shows the time but the colors are completely off. Any help would be appreciated.

**Hardware you are using**
-Raspberry Pi 3 B+
- The Matrix driver (Adafruit HAT)
- No Flicker Mod? 

**Software**
- Which version of the software are you on? 1.6.5
[image]
[image]

riffnshred, 2021-08-28:
have you made the anti-flicker mod (solder a jumber between GPIO 4 and 18) ?

if so try this:

`sudo python3 src/main.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=50 --led-slowdown-gpio=2`

kempdawg20, 2021-08-28:
> have you made the anti-flicker mod (solder a jumber between GPIO 4 and 18) ?
> 
> if so try this:
> 
> `sudo python3 src/main.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=50 --led-slowdown-gpio=2`
> 
> EDIT: If you unplugger the ribbons in the process of changing raspberry pi, make sure they are properly set in the socket. Also play with them while the screen is running, might be your ribbon cable failing (but I doubt it)

Riff, haven't done the ant-flicker mod. Played around with the ribbon and no go. I've added some photos of the pi to show the build if that helps at all. Feel like I have good separation from everything.
[image]
[image]
[image]

kempdawg20, 2021-08-28:
> have you made the anti-flicker mod (solder a jumber between GPIO 4 and 18) ?
>
> if so try this:
>
> `sudo python3 src/main.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat-pwm [...]

RIFF, I'm an idiot, was powering through the USB and not the hat. (bangs head against desk)

riffnshred, 2021-08-28:
All looks good. you said that GPIO slowdown at 4 shows time but the colors are off. Did you get your RGB matrix from Adafruit or Aliexpress ?

closed 2021-08-28 by kempdawg20

riffnshred, 2021-08-28:
ROFL !!!!!! 

Awesome ! enjoy the now standard speed hah

kempdawg20, 2021-08-28:
> ROFL !!!!!!
> 
> Awesome ! enjoy the now standard speed hah

Getting some ghosting but that I can figure out on my own lol. Was about to spend more money on more stuff lol!

riffnshred, 2021-08-28:
Yah check out https://github.com/hzeller/rpi-rgb-led-matrix, you will find tags to play with in your command to optimise de display. I strongly recommend the antiflicker mod if you have a soldering iron. they are cheap these days.