riffnshred/nhl-led-scoreboard #308: Pi3A+ Power Supply

dawelge01, 2021-09-17:
If you use the power supply that plugs into the RGB, will it still power the Pi 3A+ board?

pjockey, 2021-09-17:
I have powered the Hat and the Pi with the power going thru the Hat and it did work. I did have a problem with the wifi stay connected to my network and I'm not sure it it was the Pi, the Hat or the Network or perhaps the Hat touching something on the PI even with attempts to shield it.

I replaced the 3A+ with a 3B+ and still powering thru the Hat and the device has worked flawlessly and the wifi has not dropped once.

riffnshred, 2021-09-18:
I've been running on Pi 3A+ for over a year now and its powered through the Bonnet. No issue at all. Make sure you dont plug anything on the usb port as the current pulled by the raspberry pi can get real close to the limit the Bonnet or HAT let through (1Amp).

closed 2021-09-18 by riffnshred

dawelge01, 2021-10-06:
I just tested this and follow up. I have to power my bonnet and my Pi 3A+. Do I need a different size amperage?

riffnshred, 2021-10-07:
If you have a 4A psu, du not use a brightness higher than 50 and you should be fine. otherwise, you should still be ok but I do recommend 8A and over.