riffnshred/nhl-led-scoreboard #324: Getting ERROR Failed to get Goal details for current live game. will retry on data refresh

txrangersxx, 2021-10-17:
I have everything setup.  THe test scripts from the RGB matrix work fine.  
But when I run the scoreboard I get that error.  My command to start is:

sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
I used the setup to configure and this is my config.json

"debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Stars"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "team_summary",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "team_summary",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "central",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "weather": {
            "enabled": false,
            "view": "full",
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": false,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "EC",
            "update_freq": 5,
            "show_alerts": false,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": false,
            "animations": true,
            "start": "22:00",
            "stop": "08:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 7,
            "delay": 30
        },
        "dimmer": {
            "enabled": false,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 0,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""

pjockey, 2021-10-17:
What happens if you change your conference from eastern to western? Central is in the west is is not?

txrangersxx, 2021-10-17:
CHanged it still getting the error but getting data during intermission now

pjockey, 2021-10-17:
> CHanged it still getting the error but getting data during intermission now

There is a missing "," 
```
"post_game": [
"team_summary",
"scoreticker",
"clock"
]
},
"boards": {
```

After the ] bracket.

riffnshred, 2021-10-18:
does it still does it? Can you copy and paste the full log?