riffnshred/nhl-led-scoreboard #339: Getting error ImportError: cannot import name 'graphics' from 'rgbmatrix' when trying to run samples

txrangersxx, 2021-11-09:
I had to rebuild everything from scratch due to a failed sdcard and no backup and when i try to run samples or the scoreboard I get this error I followed the steps exactly like I did before the only thing I can think of is the repository is now bullseye and not buster.  Can someone let me know what to do?

sudo python3 runtext.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-brightness=60
Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/submodules/matrix/bindings/python/samples/runtext.py", line 4, in <module>
    from rgbmatrix import graphics
ImportError: cannot import name 'graphics' from 'rgbmatrix' (/home/pi/nhl-led-scoreboard/submodules/matrix/bindings/python/rgbmatrix/__init__.py)

txrangersxx, 2021-11-12:
Went back to buster and now getting this error.  Anyone around to look at this?

Traceback (most recent call last):
  File "runtext.py", line 3, in <module>
    from samplebase import SampleBase
  File "/home/pi/nhl-led-scoreboard/submodules/matrix/bindings/python/samples/samplebase.py", line 7, in <module>
    from rgbmatrix import RGBMatrix, RGBMatrixOptions
  File "/usr/local/lib/python3.7/dist-packages/rgbmatrix/__init__.py", line 7, in <module>
    from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions
  File "core.pyx", line 5, in init rgbmatrix.core
  File "/usr/local/lib/python3.7/dist-packages/PIL/Image.py", line 114, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

cmbehan, 2021-11-13:
I'm getting the same error now.   @txrangersxx ddi you ever resolve it, and if so what did you do?

riffnshred, 2021-11-13:
try to run the install.sh script again. sometimes some required libraries don't install because of errors. It happens from time to time. The issue is probably not the version of OS you are using so I would stick to the latest.

```
chmod +x scripts/install.sh
./scripts/install.sh
```

txrangersxx, 2021-11-14:
I have tried it multiple times, I have reinstalled the whole PI from scratch with the same issue

cmbehan, 2021-11-14:
> I have tried it multiple times, I have reinstalled the whole PI from scratch with the same issue

Same here.   Happened the same way everytime on a Pi Zero 2W.     I then tried using the OS image to start from scratch and that worked just fine.     Don't know if that's an option for you, @txrangersxx, but that got me working quickly.

falkyre, 2021-11-19:
@txrangersxx  If you run `sudo apt install libopenjp2-7-dev` it should fix that error on buster.  Your initial install, do you recall what version of python3 that was installed?  There's an issue with the rgb matrix library and python 3.9.  There is a work around for it:  

`python3 -m pip install cython`
`cd path_to_repo/bindings/python/rgbmatrix/`
`python3 -m cython -3 --cplus *.pyx`

falkyre, 2021-11-19:
I was able to reproduce this with the latest raspberry pi OS version and have submitted a PR to @riffnshred to update the install.sh to fix this issue.  See #341

txrangersxx, 2021-11-23:
Seems to be fixed except for supervisor not working under buster but I suspect thats a supervisor issue and nothing to do with with the scoreboard code

closed 2021-11-23 by txrangersxx