riffnshred/nhl-led-scoreboard #343: Hi Team,

kevingc94, 2021-11-20:
**Describe the bug**
A clear and concise description of what the bug is.

**Logs**
Copy and past any error or messages that you can get. 

**Your config file**
A screen shot of your config.json file can be really useful to help you. 

**Hardware you are using**
- Raspberry pi model
- The Matrix driver (Adafruit Bonnet or Adafruit HAT)
- Anti-flicker mod or not? 

**Software**
- Which version of the software are you on?

falkyre, 2021-11-20:
Couple of questions first.  What Raspberry Pi OS are you using?  The latest, which is Bullseye?  Or the last version?  Which is Buster (or to make it easier, how did you download it?  from raspberry pi or through the raspberry pi imager)?

Second question, when did you clone the repo?

The reason I ask is that it appears that the install might have not completely properly.  And we have found some issues with the matrix library  and Python versions greater than 3.8 that we just fixed earlier tonight.

kevingc94, 2021-11-20:
Running the command Sudo Pip3 install cairosvg solved the problem

For reference:

Raspbian GNU/Linux 10 (buster)
The PI came with a SD with raspbian already installed.
I cloned yesterday around 8 pm Easter Time.
I am using Python3 3.7.3.

falkyre, 2021-11-20:
Good to hear that the sudo pip3 install worked,  everything working for you now?

kevingc94, 2021-11-21:
Right now I am just having a few issues with the display on the LED board. Trying to solve that at the moment. I can some sort of see the display, but pretty bad flickering.

I don't recall seeing issues regarding cairosvg during the install.

[image]

riffnshred, 2021-11-21:
Check your ribbon cable and make sure its well sit in the socket on both ends. Also make sure the bottom of the Bonnet or HAT does not touch the raspberry pi. Make sure its isolated with electrical tape or a few layers of kapton tape. 

If you have not dont the antiflicker mod, there will be flickering regardless

riffnshred, 2021-11-21:
Closing because the main issue is solved

closed 2021-11-21 by riffnshred