riffnshred/nhl-led-scoreboard #352: Game screen never shows

jrakestraw, 2021-11-29:
Running into an issue with showing games. I did a restart the day before the game and after 10 min into the game, still no scoreboard. I restarted the job and looked at the logs from supervisor. Looks like it's sees that it is a game day, starts the game, then quickly says it's final.

2021-11-27 20:28:57 INFO Rendering...
2021-11-27 20:28:57 INFO Game Day Wooooo
2021-11-27 20:28:57 INFO Showing Game
2021-11-27 20:28:57 INFO Checking for new day
2021-11-27 20:28:57 INFO It is not a new day
2021-11-27 20:28:57 INFO FINAL
2021-11-27 20:28:58 INFO Showing Post-Game

 This is part of the config.json file, if you need the rest, please let me know.

"debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "24h",
        "end_of_day": "23:30",
        "location": "Hilliard, Ohio",
        "live_game_refresh_rate": 10,
        "teams": [
            "Blue Jackets"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": true
        }
    },
        "states":{
                "off_day": [
                         "scoreticker",
                         "team_summary",
                         "standings",
                         "clock",
                         "weather",
                         "wxalert"
                ],
                "scheduled": [
                        "team_summary",
                        "scoreticker",
                        "clock",
                        "weather",
                        "wxalert"
                ],
                "intermission": [
                        "scoreticker",
                        "team_summary"
                ],
                "post_game": [
                        "team_summary",
                        "scoreticker",
                        "clock",
                        "weather",
                        "wxalert"
                ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "division",
            "divisions": "metropolitan",
            "conference": "eastern"
        },

I am running the most up to date version that is out there.

riffnshred, 2021-11-30:
Change your end of day time in the config to be between 00:00 and 12:00. Thats what cause your issue

jrakestraw, 2021-11-30:
Thanks. Just updated it. For the screensaver, I have it at 23:31 and it works just fine, so wasn't sure.

closed 2021-12-01 by jrakestraw