riffnshred/nhl-led-scoreboard #355: Running Multiple Boards

jeepdrvr71, 2021-12-15:
Is there a way to open multiple boards at the same time. I found an EPL scoreboard that I would like to run as well. I am in an EPL Fantasy league and would like to see the scores during game days as opposed to building a second board.

pjockey, 2021-12-16:
I would try things out by adding the Supervisor program to your Pi. https://github.com/riffnshred/nhl-led-scoreboard/wiki/Step-by-step-guide. Step 7. I switch between the NHL, MLB and the NFL. Don't think you can have any of them on the scoreboard at the same time.

necromancyr, 2022-01-05:
This can help switch between boards if you're using Supervisor and have a button as well: https://github.com/necromancyr/swap-led-scoreboard/blob/master/swapscript.py