riffnshred/nhl-led-scoreboard #360: Power Supply Issue

dawelge01, 2022-01-11:
I'm having issues with the power supply. I have 2 power supplies 1 powering the board the other powering the pi. I attached pictures to show what I have. I bought another power supply to try and power both at once. I'm not sure if I'm missing something or why it is not passing through. I have a raspberry A+.
[image]
[image]
[image]
[image]
[image]

falkyre, 2022-01-11:
When you say a pi A+, you mean the 3A+?

The new power supply, the 5V 10A, is a good sized one to power both your pi and the panel through the barrel adapter (the 5V in) on your bonnet (the 5v 4A one would work as well).  The 5V out will go to your panel.    If you are just plugging the 5V 2.5A power adapter into the micro-usb it won't power the panel, it'll only power the pi.

dawelge01, 2022-01-12:
Sean,
 Yes I meant the 3A+ model.

I plugged the new power adapter into the 5v barrel on the bonnet and it doesn’t power the pi. Could there be something wrong with the bonnet?

Thanks,
Danny

riffnshred, 2022-01-12:
If you take off the bonnet and you power on your raspberry pi, does it run? Like @falkyre said, you should not need to power the Pi since its a Pi3 A+. What Im wondering is if something is shorting the pi underneath it since you sticked it to the panel.

falkyre, 2022-01-12:
Looks like there's electrical tape between bonnet and pi

riffnshred, 2022-01-12:
Not sure if its related but, what have you connected to those white wires? I see one connected to ground, and one to the SCL ?

dawelge01, 2022-01-12:
> Yes, it powers on with the pi with the pi power adapter. I only hot glued the pi to the board.

dawelge01, 2022-01-12:
> Not sure if its related but, what have you connected to those white wires? I see one connected to ground, and one to the SCL ?

Yes the ground and SCL.

dawelge01, 2022-01-12:
So the 5v barrel on the bonnet should power the pi and bonnet correct? Am I missing anything else?

falkyre, 2022-01-12:
Ok so without the bonnet on your pi, your pi will power up just fine?

When you just supply power to the bonnet with it attached to the pi, does the 5V OK light on the bonnet turn on as in this picture below?

[image]

The anitflicker mod is to connect pins 4 and 18, which it looks like you have.  The white wires you have are connected to SCL and GND and then to what?  It shouldn't cause an issue as they are free pins.

dawelge01, 2022-01-12:
> Ok so without the bonnet on your pi, your pi will power up just fine?
>
> When you just supply power to the bonnet with it attached to the pi, does the 5V OK light on the bonnet turn on as in this [...]

Sorry, you are right that is not the anti flicker that is the push button to turn off. Which technically I don't need because I use the Supervisor webpage. With them both connected and powered on the 5v OK light is lit, just the pi is not.
[image]
.

falkyre, 2022-01-12:
OK,

next question.  When you power the pi up by itself, does it get an IP address and you can connect to it through SSH?  If it does, when you add the bonnet back onto the pi, does it show up on the network?  Are you basing that the pi is not working because the red led on the pi is not lit up?  I have a 3a+ that's powered up through the Adafruit HAT version for the RGB Matrix and the red power light is not lit up.

dawelge01, 2022-01-12:
When I power pi by itself it does get an IP, I did not SSH to it but Supervisor was able to connect to it. Are you saying to leave the pi powered on and plug the bonnet in or connect the bonnet and use the power off the bonnet? I powered mine through the Adafruit HAT and the red light never comes on and the either does the LED board.

falkyre, 2022-01-12:
Use the power off of the bonnet, ultimately one power supply is what you are looking for.  I'm trying to find out if the power through the bonnet is actually powering your pi as well.  Because you are expecting supervisor webpage being up, I'm assuming you can figure out how to connect to it and can find the IP address.  When you power the pi through the barrel jack on the bonnet (connected to the pi), can you find the address of the pi?  I use fing or wifiman on my phone to scan my network and look for raspberry pis to find it's IP address.

dawelge01, 2022-01-13:
> Use the power off of the bonnet, ultimately one power supply is what you are looking for. I'm trying to find out if the power through the bonnet is actually powering your pi as well. Because you are [...]

I know how to find the IP on my router and it is not showing up or passing through to the pi. I know this bonnet worked before because I replace my Raspberry pi Zero with a 3A+. Not understanding why it's not passing through.

jediempi, 2022-01-13:
Are the white cables hooked up to a button for shutting it down? Is the button activating the shutdown (on accident)? Can you plug in a monitor to troubleshoot while the bonnet is attached?

It is probably something simple. I feel for you. Best of luck. 🤦‍♂️

falkyre, 2022-01-13:
> > Use the power off of the bonnet, ultimately one power supply is what you are looking for. I'm trying to find out if the power through the bonnet is actually powering your pi as well. Because you are expecting supervisor webpage being up, I'm assuming you can figure out how to connect to it and can find the IP address. When you power the pi through the barrel jack on the bonnet (connected to the pi), can you find the address of the pi? I use fing or wifiman on my phone to scan my network and look for raspberry pis to find it's IP address.
> 
> I know how to find the IP on my router and it is not showing up or passing through to the pi. I know this bonnet worked before because I replace my Raspberry pi Zero with a 3A+. Not understanding why it's not passing through. As soon as I add the power adapter to the pi it lights up but then I have 2 plugs. Should I look into replacing the bonnet or pi?

Do you still have the pi zero?  If so, try attaching the bonnet to it and see if it still works.  If it does, then it's likely something with your 3A+ and the GPIO header I think and I'd look at getting it replaced.

riffnshred, 2022-01-13:
Also, if what @falkyre said does not work, could you show a picture of the bottom of the Bonnet and the top and bottom of the pi?

riffnshred, 2022-01-13:
at this point, you probably are going to find better support on the adafruit forum or discord.

dawelge01, 2022-01-13:
> at this point, you probably are going to find better support on the adafruit forum or discord.

I will get those pictures over to you guys and test my other pi. This is really strange I was wondering if there is a code/feature I need to turn on in the pi settings. Let me see if I can get another pi 3A+. I really appreciate everyone's help, this really has me baffled.

dawelge01, 2022-01-14:
Well after messing with the bonnet on my other pi I think it finally went out. It started flickering and the 5v is dim. Looks like my bonnet died. I can’t seem to find one at the time.

closed 2022-01-14 by riffnshred

dawelge01, 2022-01-14:
I’m not big on solder because I have issues making the solder stay. Do you have any suggestions on what kind of solder to use?

Closed #360<https://github.com/riffnshred/nhl-led-scoreboard/issues/360>.

riffnshred, 2022-01-14:
Its all about the technic. If it does not stick its because the area you are trying to stick it too is not hot enough. I use some real cheap solder from amazon usually.  the size I like is 0.8mm diameter. bigger then that I find its hard to control

[something like this](https://www.amazon.com/TOWOT-Purity-Electrical-Soldering-Content/dp/B09KM31DQ9/ref=sr_1_1_sspa?crid=1DB0KRHLXM5U0&keywords=solder&qid=1642127507&sprefix=solder%2Caps%2C63&sr=8-1-spons&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzN01NVlpNSlZDT1IwJmVuY3J5cHRlZElkPUEwMTg3NzU0M0JSVFgwVjJWUkc5RiZlbmNyeXB0ZWRBZElkPUEwMjY3ODE5U0xEQUdVQ1ZJVEVUJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ&th=1)