riffnshred/nhl-led-scoreboard #361: Issues Running Final Display

Kanuck89, 2022-01-13:
Hi,

I have followed through the set up instructions for the first time and am having issues with the final display. I am able to get "Hello world!" to scroll on the screen. and my config.json file passes the nhl setup validation.

However, when I go to run it through Supervisor, I receive a 'fatal' state with a description of 'exited too quickly'. Checking the logs on supervisor I see the comment of "python3: can't open file '/home/pi/nhl-led-scoreboard/runtext.py': [Errno 2] No such file or directory".

[image]

Any thoughts on what could be the issue? I'm rather new to this so I'm sure more information is needed to help describe the issue. 

Thanks

falkyre, 2022-01-13:
If you are getting fatal errors in supervisor, the best first step would be to not use supervisor.  The tail -f stderr doesn't always work for some reason.  Take what your command line is for supervisor and run the command in a terminal.  Then you'll get the output live and when it crashes, you'll see exactly what error causes the crash.   Once you get that, reply to this thread and then @riffnshred or myself can figure out where things are crashing.

Kanuck89, 2022-01-13:
Thanks for the feedback.

Executing the command line in a ssh terminal shows the LED display and cycles through the different scoreboards. However only if I'm in the 'nhl-led-scoreboard' directory as shown in the first screenshot. From there I checked the directory in the scoreboard.conf file, but I believe everything is as it should be? 

[image]

[image]

Kanuck89, 2022-01-13:
After rebooting again appears I can now control and launch it from supervisor. Not sure why it wasn't working previously after the command line update as I went through this process a number of times with no success. 

Anyways, appreciate the quick feedback and support. I'll go ahead and close this issue with this comment.

closed 2022-01-13 by Kanuck89

riffnshred, 2022-01-13:
tis the motto of all programmer: "It works now, I don't know why ¯\_(ツ)_/¯"