riffnshred/nhl-led-scoreboard #365: Led Matrix issues

daz22, 2022-01-25:
**Describe the bug**
Follow instruction and everything seems to be working but issues with the display. I have two RGB LED Matrix boards and i'm getting this same issue with both boards. 

Pi3 
Adafruit Hat w/RTC 
no anti flicker 
[image]

pjockey, 2022-01-25:
How are you powering the unit? I see a barrel jack lying on the carpet. You should power via the Adafruit Hat, not the Pi.

daz22, 2022-01-25:
I have 5v 4A going to the hat and the Cana power supply for the pi

daz22, 2022-01-25:
[image]

pjockey, 2022-01-25:
OK. I thought you should be able to power up everything by powering just the Hat. 

You should try the anti flicker. I makes a world of difference. Do you have a breadboard wire you can place in the Hat to try the anti flicker?

riffnshred, 2022-01-25:
those missing lines are caused by a miss connection of one of the pins of the data port or cable or miss connection of one the pins that connect the HAT to the raspberry pi. If you have one, use your multimeter in the Continuity mode to probe where the missing connection is.

daz22, 2022-01-25:
Ok! I will try that!

daz22, 2022-01-25:
Check all the pins good continuity to all the pins.

riffnshred, 2022-01-26:
So it could also be a faulty ribbon cable. if you play with it while the screen run, do you see something happen ? did you buy the panel also from Adafruit? or somewhere else. at the point I would go on their forum or discord for help as your problem is not related to the nhl-scoreboard software.

Brune04, 2022-11-13:
Did you ever figure this out? I'm having the exact same issue

daz22, 2022-11-13:
@Brune04

closed 2022-11-13 by daz22