riffnshred/nhl-led-scoreboard #372: Error during nhl_setup file Not Found  32 error on 64 bit system

Overboard-code, 2022-05-04:
When I try to run ./nhl_setup on my 64bit Raspi OS on a  Pi 3B  I see:
```
pi@hockey:~/nhl-led-scoreboard $ ./nhl_setup
-bash: ./nhl_setup: No such file or directory
```
I look ate the executable and it is 32bit 
```
pi@hockey:~/nhl-led-scoreboard $ file nhl_setup
nhl_setup: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=ad32521ad141d04ca4fc066798301621367c7964, stripped
```
My system is 64bit
```
pi@hockey:~/nhl-led-scoreboard $ uname -a
Linux hockey 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```
I tried setting up for 64 bit cross executions:
```
pi@hockey:~/nhl-led-scoreboard $ ./nhl_setup
-bash: ./nhl_setup: No such file or directory
pi@hockey:~/nhl-led-scoreboard $ sudo ln -s /usr/arm-linux-gnueabihf/lib/ld-linux-armhf.so.3 /usr/lib/ld-linux-armhf.so.3
pi@hockey:~/nhl-led-scoreboard $ ./nhl_setup
./nhl_setup: error while loading shared libraries: libdl.so.2: cannot open shared object file: No such file or directory  
```
I see libdl.so.2 in the arm 32 library
```
pi@hockey:/lib $ ls  /usr/arm-linux-gnueabihf/lib/
ld-2.31.so           libBrokenLocale-2.31.so  libdl.so.2      libnsl.so.1            libnss_files-2.31.so   libpthread-2.31.so  librt.so.1           libutil.so.1
ld-linux-armhf.so.3  libBrokenLocale.so.1     libm-2.31.so    libnss_compat-2.31.so  libnss_files.so.2      libpthread.so.0     libSegFault.so
ld-linux.so.3        libc-2.31.so             libmemusage.so  libnss_compat.so.2     libnss_hesiod-2.31.so  libresolv-2.31.so   libthread_db-1.0.so
libanl-2.31.so       libc.so.6                libm.so.6       libnss_dns-2.31.so     libnss_hesiod.so.2     libresolv.so.2      libthread_db.so.1
libanl.so.1          libdl-2.31.so            libnsl-2.31.so  libnss_dns.so.2        libpcprofile.so        librt-2.31.so       libutil-2.31.so
```
Is there a way to do this on a 64 bit system?  Have I just missed some big obvious thing?

riffnshred, 2022-05-04:
I have not tested the 64bit version of Pi OS sadly and I dont intend on supporting it as it does not bring any improvement for this project. I recommend sticking with the 32 bit version for now.

closed 2022-05-04 by riffnshred

Overboard-code, 2022-05-04:
OK Will do  I am early days with the install  Just got the
panel working and playing with Demos
Thanks so much for the nice program