riffnshred/nhl-led-scoreboard #373: Started scoreboard and see: "The 20212022 Season playoff has not started yet or is unavailable"

Overboard-code, 2022-05-05:
**Describe the bug**
A clear and concise description of what the bug is.
When I start the scoreboard I see:
The 20212022 Season playoff has not started yet or is unavailable
 and on my LED panel I just see the clock and weather.  I don't see anything else.  I expected to see  Conference standings etc.  

**Logs**
Copy and past any error or messages that you can get. 
```
pi@hockey:~/prem-led-scoreboard $ sudo python3 src/main.py --led-rows=64 --led-cols=128 --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=2
INFO: Validating config.json.....
INFO: config.json passes validation
The 20212022 Season playoff has not started yet or is unavailable
/usr/local/lib/python3.9/dist-packages/apscheduler/util.py:436: PytzUsageWarning: The localize method is no longer necessary, as this time zone supports the fold attribute (PEP 495). For more details on migrating to a PEP 495-compliant implementation, see https://pytz-deprecation-shim.readthedocs.io/en/latest/migration.html
  return tzinfo.localize(dt)
```
 
**Your config file**
```
{
    "debug": false,
    "loglevel": "WARNING",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "Winston-Salem, nc",
        "live_game_refresh_rate": 10,
        "teams": [
            "Hurricanes",
            "Red Wings"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
                "pref_team_only": true
        }
    },
    "states": {
        "off_day": [
            "clock",
            "weather",
            "wxalert",
            "seriesticker",
            "standings",
            "team_summary",
            "stanley_cup_champions"
        ],
        "scheduled": [
            "clock",
            "weather",
            "wxalert",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary",
            "stanley_cup_champions"
        ],
        "intermission": [
            "clock",
            "weather",
            "wxalert",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary"
        ],
        "post_game": [
            "clock",
            "weather",
            "wxalert",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary",
            "stanley_cup_champions"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": false,
            "standing_type": "division",
            "divisions": "metropolitan",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": false
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": false,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": true,
            "view": "full",
            "units": "imperial",
            "duration": 60,
            "data_feed": "OWM",
            "owm_apikey": "584a45a2e53805b06ec4bb1ac18680ab",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "NWS",
            "update_freq": 5,
            "show_alerts": true,
            "nws_show_expire": true,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": true,
            "animations": true,
            "start": "22:00",
            "stop": "06:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 24,
            "delay": 30
        },
        "dimmer": {
            "enabled": true,
            "source": "software",
            "daytime": "",
            "nighttime": "",
            "offset": 50,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}
```
**Hardware you are using**
- Raspberry pi model 3B 
- The Matrix driver  Adafruit Bonnet 
- Anti-flicker mod  is on and works
- Panel is this one from Amazon: https://www.amazon.com/gp/product/B0993PMQLY/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1 
       P2.5 Indoor RGB Led Matrix Panel Board, 12864 Pixels, 2.5mm Pitch, 1/32 Scan, Full Color HD Video Wall LED Display Module (RGB, 320x160mm)
I start the scoreboard with 
```
sudo python3 src/main.py --led-rows=64 --led-cols=128 --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=2
```
**Software**
- Which version of the software are you on? 1.5.4

Overboard-code, 2022-05-05:
So problem remains that scoreboard does not know it's playoffs.  Also i don't see standings or team stats for my teams

Overboard-code, 2022-05-05:
Side question  In config/layouts I see layout_128x64.json  Can I use this?

riffnshred, 2022-05-05:
Neat setup with the radio !

You are using an older version and a bunch of stuff wont work on the version you are using. 

Also, I don't support anything other then 64x32 right now but yes you can create a layout_128x64.json to create your own layout and stuff. but again, some stuff will still be weird.

Overboard-code, 2022-05-05:
Cool on the layout  How do I use a new layout?  I can accept some weirdness

Overboard-code, 2022-05-06:
Got new version  I did a default config for just one team and it worked  
Thanks

closed 2022-06-24 by riffnshred