riffnshred/nhl-led-scoreboard #374: Error after nhl-setup when not simple setup

Overboard-code, 2022-05-06:
after trying simple config I wanted to add weather and other screens results were:
```

pi@hockey:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-rows=64 --led-cols=128 --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=2
INFO: Validating config.json.....
WARN: config.json fails validation: error: [data.preferences.goal_animations must contain ['pref_team_only'] properties]
WARN: Rerun the nhl_setup app to create a valid config.json

```
In the config file it looks like:
```
     "goal_animations": {}
```
it looks like goal animations didn't configure correctly:  I think it should be 
```
       "goal_animations": 
        {
                "pref_team_only": false
        }

```
**Logs**
```
pi@hockey:~/nhl-led-scoreboard $ ./nhl_setup
   _  ____ ____     __   _______
  / |/ / // / /    / /  / __/ _ \
 /    / _  / /__  / /__/ _// // /
/_/|_/_//_/____/ /____/___/____/

   _____________  ___  _______  ____  ___   ___  ___
  / __/ ___/ __ \/ _ \/ __/ _ )/ __ \/ _ | / _ \/ _ \
 _\ \/ /__/ /_/ / , _/ _// _  / /_/ / __ |/ , _/ // /
/___/\___/\____/_/|_/___/____/\____/_/ |_/_/|_/____/

   ________________  _____
  / __/ __/_  __/ / / / _ \
 _\ \/ _/  / / / /_/ / ___/
/___/___/ /_/  \____/_/

nhl_setup V1.6.2
nhl led scoreboard V1.6.9
Now validating config......
Your config.json passes validation and can be used with nhl led scoreboard
🥅🏒🚨 Do you see a net,stick and horn? Yes
🥅 Do you want a simple default setup with one team selection (Y)? No
🥅 What kind of setup do you want? full
🥅 Do you want full debug on (only use this if asked to)? (outputs ALL log messages to console) No
🥅 What log level do you want?  INFO
🥅 Do you want live_mode on? (Shows live game data of favorite team) Yes
🥅 Select the time format 12h
🥅 When do you want end of day to be? 22:00
🥅 Your location to override latitude and longitude lookup via IP (City, State or City, Province or lat,lon or home address)
🥅 Live game refresh rate? (rate at which a live game will call the NHL API to catch the new data, 10 seconds is the lowest) 10
🥅 Select your preferred team: (Use arrow keys)
 » Hurricanes
   Avalanche
   Blackhawks
   Blues
   Blue Jackets
   Bruins
   Canadiens
   Canucks
   Capitals
   Coyotes
   Devils
   Ducks
   Flames
   Flyers
   Golden Knights
   Islanders
[... 242 lines cut ...]
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}

```
**Hardware you are using**
- Raspberry pi model
- The Matrix driver (Adafruit Bonnet or Adafruit HAT)
- Anti-flicker mod or not? 

**Software**
- Which version of the software are you on?

falkyre, 2022-05-06:
I just downloaded a fresh install of the latest repository (1.6.9) so no config.json was included and I had to run the nhl_setup.  Ran simple setup and the goal_animations has the pref_teams_only: true.

I see in another issue you raised that you were running an older version of the scoreboard.  Did you create the simple config with that version?

Overboard-code, 2022-05-06:
No. I deleted the whole directory tree. Then did a 'git' then the config
simple.  Then config again. I expected'false' as the value the simple
config had 'true' for me as well.

falkyre, 2022-05-06:
So when you redid the setup, you selected false for the goal animations?  And it didn't write anything back to the config ?

Overboard-code, 2022-05-06:
Correct.

falkyre, 2022-05-07:
Tried 3 times, can't recreate the issue you are seeing.  What I do see from your run of the nhl_setup above is that it looks like the prompt specific to the goal animations did not render correctly.

🥅 Do you want goal animations for only preferred team?

There should be an answer (Yes or No) no matter what (it will default to whatever it pulled from your original config.json)

At this point, I'd chalk it up to one of those un reproducible things.  Are you able to reproduce the issue consistently?

Overboard-code, 2022-05-11:
Yes I agree.  I ran nhl-setup again and saw:
`🥅 Do you want goal animations for only preferred team? No`
And it worked fin in the config:
```
      "goal_animations": {
            "pref_team_only": false
        }
```
Thanks so much for your time.

Overboard-code, 2022-05-11:
Found it.  If I just hit Enter instead of 'Y' or 'N' for goal animations it fails  If I enter 'Y' or 'N' on the keyboard it works.  I tried it three times  with the same results.
Hit N and I see:
`🥅 Do you want goal animations for only preferred team? No`
Hit Enter and I see:
`🥅 Do you want goal animations for only preferred team?`

falkyre, 2022-05-11:
OK,  was able to recreate and fix the problem.  Thanks for finding it.  I'll add the fix and the new binary in a PR for @riffnshred to add to the next release.

Overboard-code, 2022-05-11:
Thanks so much!  I really appreciate what you are doing.

falkyre, 2022-05-14:
This is fixed in #377 

@riffnshred will just need to merge it for the next release,

falkyre, 2022-06-24:
@riffnshred this can be closed now.  It's part of the 1.6.10 version you pushed out last week.

closed 2022-06-24 by riffnshred