riffnshred/nhl-led-scoreboard #379: Board Restarts during Post Game SC Champions Check

sflems, 2022-06-16:
**Describe the bug**
Board Restarts during Post Game SC Champions Check

**Logs**
```
2-06-15 23:03:40 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80
22-06-15 23:03:40 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/schedule?date=2022-6-15&expand=schedule.linescore HTTP/1.1" 200 None
22-06-15 23:03:40 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80
22-06-15 23:03:40 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2021030411/feed/live?site=en_nhl HTTP/1.1" 200 None
22-06-15 23:03:41 DEBUG STREAM b'IHDR' 16 13
22-06-15 23:03:41 DEBUG STREAM b'IDAT' 41 3003
22-06-15 23:03:41 DEBUG STREAM b'IHDR' 16 13
22-06-15 23:03:41 DEBUG STREAM b'IDAT' 41 1630
22-06-15 23:03:41 DEBUG Starting new HTTP connection (1): statsapi.web.nhl.com:80
22-06-15 23:03:41 DEBUG http://statsapi.web.nhl.com:80 "GET /api/v1/game/2021030411/feed/live?site=en_nhl HTTP/1.1" 200 None
22-06-15 23:03:41 DEBUG STREAM b'IHDR' 16 13
22-06-15 23:03:41 DEBUG STREAM b'tEXt' 41 25
22-06-15 23:03:41 DEBUG STREAM b'iTXt' 78 883
22-06-15 23:03:41 DEBUG STREAM b'IDAT' 973 291
INFO: Validating config.json.....
INFO: config.json passes validation
22-06-15 23:03:58 INFO Scheduler started
22-06-15 23:03:58 DEBUG Looking for jobs to run
22-06-15 23:03:58 DEBUG No jobs; waiting until a job is added
22-06-15 23:03:58 INFO Added job "UpdateChecker.CheckForUpdate" to job store "default"
22-06-15 23:03:58 INFO Using cache directory: /root/.cache/lastversion.
22-06-15 23:03:58 INFO Created instance of GitHubRepoSession
22-06-15 23:03:58 DEBUG Looking for jobs to run
22-06-15 23:03:58 DEBUG Next wakeup is due at 2022-06-16 03:00:00-07:00 (in 14161.535157 seconds)
22-06-15 23:03:58 DEBUG Looking up "https://github.com/riffnshred/nhl-led-scoreboard/releases.atom" in the cache
22-06-15 23:03:58 DEBUG Current age based on date: 1254
22-06-15 23:03:58 DEBUG Freshness lifetime from max-age: 0
22-06-15 23:03:58 DEBUG Starting new HTTPS connection (1): github.com:443
22-06-15 23:03:58 DEBUG https://github.com:443 "GET /riffnshred/nhl-led-scoreboard/releases.atom HTTP/1.1" 304 0
22-06-15 23:03:58 INFO Got HTTP status code 200 from https://github.com/riffnshred/nhl-led-scoreboard/releases.atom
22-06-15 23:03:58 INFO Checking tag v1.6.9
22-06-15 23:03:58 INFO Sanitizing string v1.6.9 as a satisfying version.
22-06-15 23:03:58 INFO Parsed as Version OK
22-06-15 23:03:58 INFO String representation of version is 1.6.9.
22-06-15 23:03:58 DEBUG Looking up "https://api.github.com/repos/riffnshred/nhl-led-scoreboard/releases/tags/v1.6.9" in the cache
22-06-15 23:03:58 DEBUG Current age based on date: 1254
22-06-15 23:03:58 DEBUG Freshness lifetime from max-age: 60
22-06-15 23:03:58 DEBUG Starting new HTTPS connection (1): api.github.com:443
22-06-15 23:03:59 DEBUG https://api.github.com:443 "GET /repos/riffnshred/nhl-led-scoreboard/releases/tags/v1.6.9 HTTP/1.1" 304 0
22-06-15 23:03:59 INFO Got HTTP status code 200 from https://api.github.com/repos/riffnshred/nhl-led-scoreboard/releases/tags/v1.6.9
22-06-15 23:03:59 INFO Got formal release for tag v1.6.9
22-06-15 23:03:59 INFO Selected version as current selection: 1.6.9.
22-06-15 23:03:59 INFO Checking tag v1.6.8
22-06-15 23:03:59 INFO Sanitizing string v1.6.8 as a satisfying version.
22-06-15 23:03:59 INFO Parsed as Version OK
22-06-15 23:03:59 INFO String representation of version is 1.6.8.
22-06-15 23:03:59 INFO Tag v1.6.8 does not contain newer version than we already found
22-06-15 23:03:59 INFO Checking tag v1.6.7
22-06-15 23:03:59 INFO Sanitizing string v1.6.7 as a satisfying version.
22-06-15 23:03:59 INFO Parsed as Version OK
22-06-15 23:03:59 INFO String representation of version is 1.6.7.
22-06-15 23:03:59 INFO Tag v1.6.7 does not contain newer version than we already found
22-06-15 23:03:59 INFO Checking tag v1.6.6
22-06-15 23:03:59 INFO Sanitizing string v1.6.6 as a satisfying version.
22-06-15 23:03:59 INFO Parsed as Version OK
22-06-15 23:03:59 INFO String representation of version is 1.6.6.
22-06-15 23:03:59 INFO Tag v1.6.6 does not contain newer version than we already found
22-06-15 23:03:59 INFO Checking tag v1.6.5
22-06-15 23:03:59 INFO Sanitizing string v1.6.5 as a satisfying version.
22-06-15 23:03:59 INFO Parsed as Version OK
22-06-15 23:03:59 INFO String representation of version is 1.6.5.
22-06-15 23:03:59 INFO Tag v1.6.5 does not contain newer version than we already found
22-06-15 23:03:59 INFO Checking tag v1.6.4
22-06-15 23:03:59 INFO Sanitizing string v1.6.4 as a satisfying version.
22-06-15 23:03:59 INFO Parsed as Version OK
[... 152 lines cut ...]
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}
```

**Hardware you are using**
- Raspberry pi model - 4 (4GB)
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) - BONNET
- Anti-flicker mod or not? - YES

**Software**
- Which version of the software are you on?
_1.6.9_

riffnshred, 2022-06-16:
Thanks !! Im on it !

riffnshred, 2022-06-24:
this is fixed on 1.6.10

closed 2022-06-24 by riffnshred