riffnshred/nhl-led-scoreboard #388: Tzdata issue

pjockey, 2022-10-03:
**Describe the bug**
pi@Scoreboard:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-cols=128 --led-rows=64 --led-panel-type=FM6126B --led-rgb-sequence=BGR
INFO: Validating config.json.....
INFO: config.json passes validation
22-10-03 17:31:33 INFO Scheduler started
22-10-03 17:31:33 DEBUG Looking for jobs to run
22-10-03 17:31:33 DEBUG No jobs; waiting until a job is added
22-10-03 17:31:33 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
22-10-03 17:31:33 DEBUG https://dd.weather.gc.ca:443 "GET /citypage_weather/docs/site_list_en.csv HTTP/1.1" 200 32901
22-10-03 17:31:37 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
22-10-03 17:31:37 DEBUG https://dd.weather.gc.ca:443 "GET /citypage_weather/xml/BC/s0000741_e.xml HTTP/1.1" 200 2515
22-10-03 17:31:37 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
22-10-03 17:31:37 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/doc/AQHI_XML_File_List.xml HTTP/1.1" 200 6315
22-10-03 17:31:38 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
22-10-03 17:31:38 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/pyr/observation/realtime/xml/AQ_OBS_JAIQY_CURRENT.xml HTTP/1.1" 200 405
22-10-03 17:31:38 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
22-10-03 17:31:39 DEBUG https://dd.weather.gc.ca:443 "GET /air_quality/aqhi/pyr/forecast/realtime/xml/AQ_FCST_JAIQY_CURRENT.xml HTTP/1.1" 200 737
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 901, in _create_plugin_instance
    plugin_cls = class_container[alias]
KeyError: 'interval'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "src/main.py", line 156, in <module>
    run()
  File "src/main.py", line 107, in run
    ecWxWorker(data,scheduler)
  File "/home/pi/nhl-led-scoreboard/src/api/weather/ecWeather.py", line 16, in __init__
    scheduler.add_job(self.getWeather, 'interval', minutes=self.weather_frequency,jitter=60,id='ecWeather')
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 425, in add_job
    'trigger': self._create_trigger(trigger, trigger_args),
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 926, in _create_trigger
    return self._create_plugin_instance('trigger', trigger, trigger_args)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 904, in _create_plugin_instance
    plugin_cls = class_container[alias] = plugin_container[alias].load()
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 2470, in load
    self.require(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 2493, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 800, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (tzlocal 4.2 (/usr/local/lib/python3.7/dist-packages), Requirement.parse('tzlocal~=2.0'))

**Your config file**
A screen shot of your config.json file can be really useful to help you. 

**Hardware you are using**
- Raspberry pi model: 3B
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) Adafruit Bonnet
- Anti-flicker mod or not? Mod

**Software**
- Which version of the software are you on? Buster/Ver 1.6.10

falkyre, 2022-10-04:
I recall coming up against this last year while developing some new functionality.  It seems there's an issue with apscheduler when the OS updates the tzlocal library or python 3.7.  What does this give you:

`pip3 show apscheduler`

and

`pip3 show tzlocal`

pjockey, 2022-10-04:
```
pi@Scoreboard:~ $ pip3 show apscheduler
Name: APScheduler
Version: 3.7.0
Summary: In-process task scheduler with Cron-like capabilities
Home-page: https://github.com/agronholm/apscheduler
Author: Alex Grönholm
Author-email: apscheduler@nextday.fi
License: MIT
Location: /usr/local/lib/python3.7/dist-packages
Requires: pytz, setuptools, six, tzlocal
Required-by: 
pi@Scoreboard:~ $ pip3 show tzlocal
Name: tzlocal
Version: 4.2
Summary: tzinfo object for the local timezone
Home-page: UNKNOWN
Author: Lennart Regebro
Author-email: regebro@gmail.com
License: MIT
Location: /usr/local/lib/python3.7/dist-packages
Requires: backports.zoneinfo, pytz-deprecation-shim
Required-by: APScheduler
```

pjockey, 2022-10-04:
`sudo apt upgrade` showed an update to `tzlocal`

pjockey, 2022-10-04:
I uninstalled and the reinstalled the tzlocal package using pip3 and received this"

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
apscheduler 3.7.0 requires tzlocal~=2.0, but you have tzlocal 4.2 which is incompatible.
```

falkyre, 2022-10-04:
Try updating apscheduler to the last V3.x version.  Don't upgrade to the V4 beta as it will break everything.

`sudo pip3 install apscheduler=3.9.1`

pjockey, 2022-10-04:
This little mess up has taken down all of my boards.  Nothing shows on the matrix, on any of the boards.

```
pi@Scoreboard:~/mlb-led-scoreboard $ pip3 show tzlocal
Name: tzlocal
Version: 4.2
Summary: tzinfo object for the local timezone
Home-page: UNKNOWN
Author: Lennart Regebro
Author-email: regebro@gmail.com
License: MIT
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: backports.zoneinfo, pytz-deprecation-shim
Required-by: APScheduler
```
```
pi@Scoreboard:~/mlb-led-scoreboard $ pip3 show apscheduler
Name: APScheduler
Version: 3.9.1
Summary: In-process task scheduler with Cron-like capabilities
Home-page: https://github.com/agronholm/apscheduler
Author: Alex Grönholm
Author-email: apscheduler@nextday.fi
License: MIT
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: pytz, setuptools, six, tzlocal
Required-by: 
pi@Scoreboard:~/mlb-led-scoreboard $ 
```

falkyre, 2022-10-04:
Not sure what to say, I'm running multiple boards (7 or 8) and various different OSes (raspbian buster , raspbian bullseye, Alpine, Diet Pi) with the tzlocal at 4.2 and apscheduler at V3.6.3 and also 3.9.1.  I'm also running different versions of Python (3.7.3 to 3.10).  If you have a spare card, I'd start with a fresh install of the OS and start the install of the boards again.  I don't run anything other than the NHL board so I don't know if you are clobbering what the NHL installs or vice versa.

pjockey, 2022-10-04:
This is the error I get now with the starting command:

```
pi@Scoreboard:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-cols=128 --led-rows=64 --led-rgb-sequence=BGR --led-row-addr-type=3
INFO: Validating config.json.....
INFO: config.json passes validation
22-10-04 17:28:00 INFO Scheduler started
22-10-04 17:28:00 DEBUG Looking for jobs to run
22-10-04 17:28:00 DEBUG No jobs; waiting until a job is added
22-10-04 17:28:00 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
22-10-04 17:28:01 DEBUG https://dd.weather.gc.ca:443 "GET /citypage_weather/docs/site_list_en.csv HTTP/1.1" 200 32894
22-10-04 17:28:04 DEBUG Starting new HTTPS connection (1): dd.weather.gc.ca:443
Unable to connect to EC, try running again in a few minutes
pi@Scoreboard:~/nhl-led-scoreboard $ 
```

falkyre, 2022-10-05:
Totally different issue.  So the other stuff with tzlocal and apscheduler are OK?  To test, update your config to not use weather, weather alerts or weather forecast and see if it works.  If it does, close this issue and open another one for the Environment Canada stuff.  I'm seeing the same issue on some of my boards right now.  Not sure if it's an EC datamart issue or something else.

pjockey, 2022-10-05:
It appears another reboot seem to have fixed whatever was the problem.

closed 2022-10-05 by pjockey

falkyre, 2022-10-05:
There was an issue with the EC Data Mart.  It was extremely slow hence the issue you were seeing.  It does look like it's recovered now.