riffnshred/nhl-led-scoreboard #390: Unable to connect to EC

pjockey, 2022-10-05:
**Describe the bug**
The scoreboard starts, but then stops, matrix has nothing on it.

**Logs**
Unable to connect to EC, try running again in a few minutes

**Your config file**
        "weather": {
            "enabled": true,
            "view": "full",
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled":false,
            "forecast_days": 3,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "EC",
            "update_freq": 5,
            "show_alerts": false,
            "nws_show_expire": false,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
**Hardware you are using**
- Raspberry pi model 3B+
- The Matrix driver (Adafruit Bonnet or Adafruit HAT): Bonnet
- Anti-flicker mod or not? : YES

**Software**
- Which version of the software are you on? : 1.6.10

falkyre, 2022-10-05:
There was an issue with the EC Data Mart.  It was extremely slow, even on a browser (taking 7 to 10 seconds to download the xml file).  It looks like this has now cleared up.  I'll look at adding something to bypass if this happens in the future and allow the board to run ion start rather than stopping.

falkyre, 2022-10-15:
This will be fixed in next release so if 9n start l,, the EC thread won't stop the board from running.

abec427, 2022-11-03:
I have the same issue even though I have owm weather selected instead of ec. The board will not start.

2022-11-03 08:15:48 ERROR Unable to connect to EC, try running again in a few minutes

falkyre, 2022-11-03:
Do you have the weather alerts for EC selected?  If so, eventhough you are using OWM, EC will be used to get the alerts.  Easiest fix until the new version comes out is to disable the weather alerts in your config.json.

abec427, 2022-11-19:
I disabled the weather boards in config.json.  It works well like that.

VegasPrinted, 2022-11-28:
Verified again today there is an issue connecting to EC

pjockey, 2022-12-22:
Is this still not working? Or can I re-enable it?
Thanks
Scott

falkyre, 2022-12-24:
If you aren't using the beta branch, this issue is still there.  The code under beta will remove the sys.exit(0) from line 103.  If you want a quick fix, you can comment or remove that line out and this will fix the looping on start if your board is unable to reach EC or if they are having data mart issues.