riffnshred/nhl-led-scoreboard #403: LED matrix showing with light blue tint

rshadle22, 2022-12-21:
**Describe the bug**
I'm unsure if this bug is due to the code or the way the RGB bonnet is set up with the hardware. The top portion of the LED matrix is shown with a light blue tint. Many of the whites are showing as light blue and reconnecting the bonnet to the raspberry pi did not fix the issue. I can't seem to find anything online that would help resolve this issue. Came here for some assistance or direction on how to troubleshoot.

[image]

**Logs**
N/A

**Your config file**
[image]
[image]

**Hardware you are using**
- Raspberry pi 3 model A+
- The Matrix driver - Adafruit Bonnet
- Anti-flicker mod - no
- Adafruit 64x32 LED matrix

**Software**
- V 1.6.10

riffnshred, 2022-12-21:
There is a misconnection or a short somewhere. Make sure the bottom of the Bonnet does not touch anything on the raspberry pi and that the data cable is well seated in both sockets.

rshadle22, 2022-12-21:
Okay thanks! I'll try and make sure those connections are tightened properly.

rshadle22, 2023-01-01:
Forgot to update this issue. There was a short with the dashboard and I was able to fix it by making sure the bonnet didn't touch the raspberry pi. Appreciate the help!

closed 2023-01-01 by rshadle22