riffnshred/nhl-led-scoreboard #404: Continual Reloading and Not getting all Information

txrangersxx, 2023-01-21:
Started a couple days ago.  THe continual reloading and now it is not getting all the information when it tries to connect.
It gets game information when games are going but in between it doesn't get updated states or last game/next game info.
Has something changed with the NHL API?

riffnshred, 2023-01-21:
are you running the latest version? I have not notice any issue recently. when you notice the issue, could you do copy the logs and create a [pastebin](https://pastebin.com/) and post it here?

nick-danger, 2023-06-29:
Having the same issue with two different installs.

stdout reports:

```
2023-06-28 22:05:06 INFO NHL-LED-SCOREBOARD - v1.6.10 (64x32)
2023-06-28 22:05:06 INFO location loaded from cache (saved -1 days ago): -redacted-
2023-06-28 22:05:06 INFO Rendering...
2023-06-28 22:05:06 INFO Your preferred teams are Off today
2023-06-28 22:05:06 INFO PING !!! Render off day
2023-06-28 22:05:06 INFO Checking for new day
2023-06-28 22:05:06 INFO It is not a new day
2023-06-28 22:05:07 ERROR Failed to get Goal details for current live game. will retry on data refresh
2023-06-28 22:05:07 ERROR Failed to get Goal details for current live game. will retry on data refresh
```

stderr reports:

```
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 156, in <module>
    run()
  File "src/main.py", line 151, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 70, in render
    self.__render_offday()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 87, in __render_offday
    self.boards._off_day(self.data, self.matrix,self.sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 81, in _off_day
    board(data, matrix,sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 241, in team_summary
    TeamSummary(data, matrix, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/boards/team_summary.py", line 74, in render
    next_game_scoreboard = Scoreboard(nhl_api.overview(next_game_id), self.data)
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 32, in overview
    return nhl_api.game.Overview(nhl_api.game.overview(game_id))
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/game.py", line 108, in overview
    id = parsed['gamePk']
KeyError: 'gamePk'
```

I injected some debugging code, and it looks like the failure occurs because nhl.com is returning this packet:

```
{'message': "Game data couldn't be found", 'messageNumber': 2}
```

when trying to retrieve the overview for game id #2023020018

I don't know the API as intimately as some, so I can't tell if this is a transient issue or not.

I also noticed that the update check was failing; performing a `sudo pip install lastversion==2.4.15` seems to have fixed that. I'm not sure how long that's been going on, but there you go.

falkyre, 2023-07-07:
If you remove the team_summary board from your off_day boards, you won't see this error.  If you keep the team_summary board in, this is fixed in release 1.6.11 :  https://github.com/riffnshred/nhl-led-scoreboard/releases/tag/v1.6.11