riffnshred/nhl-led-scoreboard #409: Frequent Crashing

harmgsn, 2023-05-23:
Last couple of weeks getting a lot of random restarts - it's becoming more and more frequent 

Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 156, in <module>
    run()
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 151, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 71, in render
    self.__render_offday()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 92, in __render_offday
    self.data.refresh_data()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 567, in refresh_data
    self.refresh_games()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 269, in refresh_games
    self.games = nhl_api.day(self.year, self.month, self.day)
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 18, in day
    return [nhl_api.game.GameScoreboard(data[x]) for x in data]
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 18, in <listcomp>
    return [nhl_api.game.GameScoreboard(data[x]) for x in data]

I'm on a Raspberry Pi 3+ with 1.6.10.

OS and all OS modules are updated as of 3/1 - crashing started about 2 weeks ago

riffnshred, 2023-05-27:
Does it still crash at the moment? can you copy and paste your config into https://pastebin.com/ and post the link here?

harmgsn, 2023-05-30:
https://pastebin.com/xbJ1FD70

According to supervisord.log:

2023-03-02 20:42:17,893 INFO exited: scoreboard (exit status 1; not expected)
2023-03-16 05:44:52,344 INFO exited: scoreboard (exit status 1; not expected)
2023-03-18 11:44:41,826 INFO exited: scoreboard (exit status 1; not expected)
2023-03-18 20:28:53,604 INFO exited: scoreboard (exit status 1; not expected)
2023-04-15 20:19:11,112 INFO exited: scoreboard (exit status 1; not expected)
2023-04-15 20:51:44,343 INFO exited: scoreboard (exit status 1; not expected)
2023-04-23 01:27:00,214 INFO exited: scoreboard (exit status 1; not expected)
2023-04-23 04:35:59,370 INFO exited: scoreboard (exit status 1; not expected)
2023-04-23 04:38:18,317 INFO exited: scoreboard (exit status 1; not expected)
2023-05-03 08:23:02,994 INFO exited: scoreboard (exit status 1; not expected)
2023-05-18 14:15:20,193 INFO exited: scoreboard (exit status 1; not expected)
2023-05-19 15:06:32,468 INFO exited: scoreboard (exit status 1; not expected)
2023-05-21 23:07:15,539 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 00:40:57,454 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 09:25:31,208 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 10:09:06,269 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 19:42:30,148 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 20:53:12,252 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:01:20,002 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:01:24,629 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:01:29,261 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:01:33,888 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:01:38,505 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:01:43,169 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:01:57,011 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:01,800 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:11,417 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:16,144 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:20,991 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:30,689 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:35,383 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:40,008 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:44,835 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:49,488 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:02:59,115 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:03,931 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:08,550 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:13,138 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:17,742 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:27,448 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:32,316 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:47,048 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:51,789 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:03:56,392 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:04:06,063 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:04:15,672 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:24:30,327 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 21:26:10,726 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 22:03:04,826 INFO exited: scoreboard (exit status 1; not expected)
2023-05-22 23:31:25,130 INFO exited: scoreboard (exit status 1; not expected)
2023-05-23 18:10:38,805 INFO exited: scoreboard (exit status 1; not expected)
2023-05-23 18:16:44,069 INFO exited: scoreboard (exit status 1; not expected)
2023-05-26 05:11:09,050 INFO exited: scoreboard (exit status 1; not expected)

falkyre, 2023-05-31:
What I would suggest is to stop the board from running in supervisor and then run it on the command line to get any errors that are spit out. Supervisor will capture the errors but sometimes it doesn't show up in the WebUI for some reason.  If you manually run on the command line, you can see right away what's causing the crash.

harmgsn, 2023-06-01:
Isn't that the crash info I provided in the initial update?  It seems to have calmed down - so it is quite possibly an API issue (no way, those never happen....... right?  /sarcasm)

falkyre, 2023-06-05:
The Pastebin you sent only had your config and the log you showed from supervisor was the supervisord and not the actual scoreboard log.  Under supervisor, the scoreboard log is named scoteboard-<random numbers>-stdout.log and there will be a matching stderr.log.  The best way to try and capture what is causing the crash is to manually run it.  Then the only logs spit out are from the scoreboard code and not anything else.  Anyways it's highly likely it was an API issue but with an error log from the scoreboard, it's just speculation at this point.

harmgsn, 2023-06-15:
2023-06-15 15:09:11 ERROR HTTPConnectionPool(host='statsapi.web.nhl.com', port=80): Max retries exceeded with url: /api/v1/schedule?date=2023-6-15&expand=schedule.linescore (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x701507c0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

Thanks for putting up with my dumb self on Discord :)  I didn't get notified of your last comment, and just kinda wrote it off...   This is DNS issue, I'm certain.  I'll deal with this on my end.  Thank you!

closed 2023-06-15 by harmgsn