riffnshred/nhl-led-scoreboard #416: Fresh install not working

ELepolt, 2023-07-27:
**Describe the bug**
It seems that python3-pillow may not be getting properly installed. Every attempt at running `sudo scripts/aptfile apt-requirements` returns this exact screenshot (note the [NEW] next to pillow)

[image]

**Logs**
Running ./script/install.sh leads to this:
```
Running rgbmatrix installation...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: cython in /home/pi/.local/lib/python3.9/site-packages (3.0.0)
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.9/site-packages/cython.py", line 16, in <module>
    from Cython.Compiler.Main import main
  File "/home/pi/.local/lib/python3.9/site-packages/Cython/Compiler/Main.py", line 26, in <module>
    from . import Errors
  File "/home/pi/.local/lib/python3.9/site-packages/Cython/Compiler/Errors.py", line 22, in <module>
    from ..Utils import open_new_file
  File "Cython/Utils.py", line 36, in init Cython.Utils
ImportError: cannot import name __version__
make -C ./lib
make[1]: Entering directory '/home/pi/nhl-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/nhl-led-scoreboard/submodules/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/home/pi/nhl-led-scoreboard/submodules/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/pi/nhl-led-scoreboard/submodules/matrix/lib'
make[2]: Leaving directory '/home/pi/nhl-led-scoreboard/submodules/matrix/lib'
make[1]: Leaving directory '/home/pi/nhl-led-scoreboard/submodules/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/pi/nhl-led-scoreboard/submodules/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/pi/nhl-led-scoreboard/submodules/matrix/lib'
make[2]: Leaving directory '/home/pi/nhl-led-scoreboard/submodules/matrix/lib'
/usr/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext
building 'core' extension
arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I../../include -I/usr/include/python3.9 -c rgbmatrix/core.cpp -o build/temp.linux-armv7l-3.9/rgbmatrix/core.o -O3 -Wall
rgbmatrix/core.cpp:1:2: error: #error Do not use this file, it is the result of a failed Cython compilation.
    1 | #error Do not use this file, it is the result of a failed Cython compilation.
      |  ^~~~~
error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
```

**Your config file**
A screen shot of your config.json file can be really useful to help you. 

**Hardware you are using**
- Raspberry pi model

Raspberry Pi 3 Model B Rev 1.2
- The Matrix driver (Adafruit Bonnet or Adafruit HAT)
Adafruit Bonnet

- Anti-flicker mod or not? 
Yes

**Software**
- Which version of the software are you on?
```
pi@scoreboard:~/nhl-led-scoreboard $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

riffnshred, 2023-07-27:
by which version of the software we mean the version of the NHL scoreboard you are running. what you posted is the version of raspberry pi OS.

We are aware that there is a new version of Pillow but it is not compatible with the current version and we force install the version that is compatible.  

when you say fresh install, do you mean you pulled the latest version.

ELepolt, 2023-07-27:
Sorry. Yes, fresh install meaning I just flashed a new SD card, and followed the step by step instructions utilizing the git pull

riffnshred, 2023-07-27:
Try installing the Raspbien Lite version of raspberry pi, the latest version of raspberry pi is not fully supported yet.

falkyre, 2023-07-27:
I think we need to update the rgbmatrix submodule to the latest version.  I'm pretty sure we can get by without forcing the cython compilation with the newest version of the submodule.  Also, the whole installation setup needs to be revisited to work properly with Bullseye (and the upcoming version Debian 12 Bookworm).  Raspberry Pi still offers the Buster install as Raspberry Pi OS (legacy).  That's what we support

falkyre, 2023-07-27:
> Try installing the Raspbien Lite version of raspberry pi, the latest version of raspberry pi is not fully supported yet.

Make sure it's the Raspbian Pi OS Lite (Legacy), that one is the Debian 10 (Buster) version

ELepolt, 2023-08-29:
Sorry to keep bugging you... I just did this again using the buster version of [Raspberry Pi OS (Legacy)](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy) and I'm still having the exact same issue as described above.

ncrowley, 2024-02-18:
I also ran into this issue, looks like whatever was causing an issue in `rgbmatrix` that required a cython rebuild is fixed. 

Commenting those lines out of the `install.sh` script fixes the issue as described.

falkyre, 2024-02-18:
Next release removes those lines in the install script.