riffnshred/nhl-led-scoreboard #418: Cannot get past loading screen

MoonMan66, 2023-07-28:
I've tried several fresh installs but no luck, it simply keeps cycling the loading screen. It used to work until i changed by SSID yesterday.

(i removed my location from the lines below but it just repeats those in the log)

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/PC/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/PC/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

2023-07-28 11:37:43   INFO NHL-LED-SCOREBOARD - v1.6.11 (64x32)
--
2023-07-28 11:37:43 INFO location loaded   from cache (saved -1 days ago): Location, CA
2023-07-28 11:37:56 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:37:56 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:38:10 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:38:10 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:38:23 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:38:23 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:38:37 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:38:37 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:38:51 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:38:51 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:39:14 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:39:14 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:39:27 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:39:27 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:39:39 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:39:39 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:39:51 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:39:51 INFO location loaded   from cache (saved -1 days ago):
2023-07-28 11:40:09 INFO   NHL-LED-SCOREBOARD - v1.6.11 (64x32)
2023-07-28 11:40:09 INFO location loaded   from cache (saved -1 days ago):

</body>

</html>

MoonMan66, 2023-07-28:
onto\n/etc/localtime is a symlink to: America/Montreal\nFix the configuration, or set the time zone in a TZ environment variable.\n'
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 156, in <module>
    run()
  File "src/main.py", line 88, in run
    scheduler = BackgroundScheduler()
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 87, in __init__
    self.configure(gconfig, **options)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 131, in configure
    self._configure(config)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/background.py", line 29, in _configure
    super(BackgroundScheduler, self)._configure(config)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 702, in _configure
    self.timezone = astimezone(config.pop('timezone', None)) or get_localzone()
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 206, in get_localzone
    _cache_tz = _get_localzone()
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 168, in _get_localzone
    tzname = _get_localzone_name(_root)
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 147, in _get_localzone_name
    raise utils.ZoneInfoNotFoundError(message)
tzlocal.utils.ZoneInfoNotFoundError: 'Multiple conflicting time zone configurations found:\n/etc/timezone: America/Toronto\n/etc/localtime is a symlink to: America/Montreal\nFix the configuration, or set the time zone in a TZ environment variable.\n'
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 156, in <module>
    run()
  File "src/main.py", line 88, in run
    scheduler = BackgroundScheduler()
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 87, in __init__
    self.configure(gconfig, **options)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 131, in configure
    self._configure(config)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/background.py", line 29, in _configure
    super(BackgroundScheduler, self)._configure(config)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 702, in _configure
    self.timezone = astimezone(config.pop('timezone', None)) or get_localzone()
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 206, in get_localzone
    _cache_tz = _get_localzone()
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 168, in _get_localzone
    tzname = _get_localzone_name(_root)
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 147, in _get_localzone_name
    raise utils.ZoneInfoNotFoundError(message)
tzlocal.utils.ZoneInfoNotFoundError: 'Multiple conflicting time zone configurations found:\n/etc/timezone: America/Toronto\n/etc/localtime is a symlink to: America/Montreal\nFix the configuration, or set the time zone in a TZ environment variable.\n'
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 156, in <module>
    run()
  File "src/main.py", line 88, in run
    scheduler = BackgroundScheduler()
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 87, in __init__
    self.configure(gconfig, **options)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 131, in configure
    self._configure(config)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/background.py", line 29, in _configure
    super(BackgroundScheduler, self)._configure(config)
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 702, in _configure
    self.timezone = astimezone(config.pop('timezone', None)) or get_localzone()
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 206, in get_localzone
    _cache_tz = _get_localzone()
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 168, in _get_localzone
    tzname = _get_localzone_name(_root)
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 147, in _get_localzone_name
    raise utils.ZoneInfoNotFoundError(message)
tzlocal.utils.ZoneInfoNotFoundError: 'Multiple conflicting time zone configurations found:\n/etc/timezone: America/Toronto\n/etc/localtime is a symlink to: America/Montreal\nFix the configuration, or set the time zone in a TZ environment variable.\n'
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
[... 634 lines cut ...]
tzlocal.utils.ZoneInfoNotFoundError: 'Multiple conflicting time zone configurations found:\n/etc/timezone: America/Toronto\n/etc/localtime is a symlink to: America/Montreal\nFix the configuration, or set the time zone in a TZ environment variable.\n'
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "src/main.py", line 156, in <module>
    run()
  File "src/main.py", line 88, in run
    scheduler = BackgroundScheduler()
  File "/usr/local/lib/python3.7/dist-packages/apscheduler/schedulers/base.py", line 87, in __init__
    self.configure(gconfig, **options)
[... cut ...]
    tzname = _get_localzone_name(_root)
  File "/usr/local/lib/python3.7/dist-packages/tzlocal/unix.py", line 147, in _get_localzone_name
    raise utils.ZoneInfoNotFoundError(message)
tzlocal.utils.ZoneInfoNotFoundError: 'Multiple conflicting time zone configurations found:\n/etc/timezone: America/Toronto\n/etc/localtime is a symlink to: America/Montreal\nFix the configuration, or set the time zone in a TZ environment variable.\n'
INFO: Validating config.json.....
INFO: config.json passes validation

MoonMan66, 2023-07-28:
seems to be the exact issue as this old thread though the same resolution seems to have had no effect.

https://github.com/riffnshred/nhl-led-scoreboard/issues/331

MoonMan66, 2023-07-28:
not sure why but after the updating pip 3 and trying these commands for the 5th time, it worked:

sudo pip3 uninstall tzlocal
sudo pip3 install tzlocal

Please archive :)

falkyre, 2023-07-28:
If you have a very old version of pip3 (as in you have never updated it in the last 18 months or so), it tends to not handle installs of newer libraries.  The tzlocal library did go through a change (as noted in #331).  I am currently working on updating the install so it works on the newer version of the Raspberry Pi OS (Bullseye) and cleaning up/updating the python libraries is one of those steps.  The tzlocal will be added to that so that the APScheduler library works cleanly.

Although looking at your error above, it appears that the /etc/timezone configuration was one of the ones that the tzlocal library was having issues with in 2021 and was fixed then (see https://github.com/regebro/tzlocal/issues/122)