riffnshred/nhl-led-scoreboard #438: NOT All Teams Working

astlgit, 2023-12-18:
Running 1.7.0, I have 3 teams that I follow and only 1 of the teams are coming up when scoreboard starts. I have The Rangers, Hurricanes and Lightning in the config.json but get an error stating that the rangers and lightning are not teams. I'm assuming that the new api have these teams names different. I've changed them to what I think they might be, i.e. New York Rangers etc but get another error that breaks the display with that these names are not in the file. Is there a way to know what the new team names are and be able to change them in the config file team name list to get the correct teams names.

**Logs**
INFO: Validating config.json.....
INFO: config.json passes validation
Rangers is not a team of the NHL. Make sure you typed team's name properly
Lightning is not a team of the NHL. Make sure you typed team's name properly

**Your config file**
"debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "location": "CARY, NC",
        "live_game_refresh_rate": 10,
        "teams": [
            "Rangers",
            "Hurricanes",
            "Lightning"

**Hardware you are using**
- Raspberry pi 4
- The Matrix driver (Adafruit Bonnet
- Anti-flicker mod -- yes

**Software**
- 1.7.0

bderickson, 2023-12-26:
I ran into a similar issue with the Blues, did a little digging, and I think I have a potential workaround. The problem seems to stem from https://github.com/riffnshred/nhl-led-scoreboard/blob/master/src/data/data.py#L488-L490, and there's a comment there describing how the current implementation won't work for teams with cities containing more than one word (like the St Louis Blues or Tampa Bay Lightning).

Here's the full list of allowed preferred teams from that method:
```
'Canucks', 'York Rangers', 'Golden Knights', 'Bruins', 'Stars',
'Avalanche', 'Jets', 'Angeles Kings', 'York Islanders', 'Maple Leafs',
'Panthers', 'Flyers', 'Capitals', 'Bay Lightning', 'Predators', 'Hurricanes',
'Jersey Devils', 'Coyotes', 'Red Wings', 'Louis Blues', 'Canadiens',
'Penguins', 'Wild', 'Flames', 'Kraken', 'Sabres', 'Oilers', 'Blue Jackets',
'Senators', 'Ducks', 'Blackhawks', 'Jose Sharks'
```

However, you can't just use the preferred names listed here because it'll fail config file verification, which relies on https://github.com/riffnshred/nhl-led-scoreboard/blob/master/config/config.schema.json#L185-L216. You'll also need to edit that list.

Once I set my preferred team to "Louis Blues" and updated the config file schema to allow that team name, things seem to have started working for me.

This is really janky and obviously not a permanent fix. I also feel like I might be missing a simpler workaround because I'm not sure how this would have worked before the NHL changed up its API. Maybe someone that knows this codebase better can chime in. Time permitting over my holiday break, I'll try to piece together a PR to fix this properly.

astlgit, 2023-12-27:
That fixed it for me, Thanks @bderickson. I knew it was something like this but just didn't know how to change things. Glad to get it back and running somewhat properly.

MordeKyle, 2023-12-30:
This totally worked for me. Changed the config.son from "Blues" to "Louis Blues" and changed the config.schema.json from "Blues" to "Louis Blues". Thanks for finding this workaround.