riffnshred/nhl-led-scoreboard #450: AttributeError: 'Data' object has no attribute 'current_round'

Overboard-code, 2024-02-29:
**Describe the bug**
I just did a clean install.  I get AttributeError: 'Data' object has no attribute 'current_round' every time and we restart

**Logs**
Loop of:
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 165, in <module>
    run()
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 160, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 80, in render
    self.__render_game_day()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 204, in __render_game_day
    self.boards._scheduled(self.data, self.matrix,self.sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 124, in _scheduled
    board(data, matrix, sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 229, in seriesticker
    Seriesticker(data, matrix, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/boards/seriesticker.py", line 26, in render
    if not self.data.current_round:
AttributeError: 'Data' object has no attribute 'current_round'
INFO: Validating config.json.....
INFO: config.json passes validation

**Your config file**
 {
    "debug": false,
    "loglevel": "INFO",
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "6:00",
        "location": "",
        "live_game_refresh_rate": 10,
        "teams": [
            "Hurricanes"
        ],
        "sog_display_frequency": 4,
        "goal_animations": {
            "pref_team_only": false
        }
    },
    "states": {
        "off_day": [
            "clock",
            "weather",
            "wxalert",
            "wxforecast",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary",
            "christmas",
            "seasoncountdown"
        ],
        "scheduled": [
            "clock",
            "weather",
            "wxalert",
            "wxforecast",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary",
            "christmas",
            "seasoncountdown",
            "stanley_cup_champions"
        ],
        "intermission": [
            "clock",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary",
            "seasoncountdown"
        ],
        "post_game": [
            "clock",
            "seriesticker",
            "standings",
            "team_summary"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 7
        },
        "seriesticker": {
            "preferred_teams_only": false,
            "rotation_rate": 7
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "conference",
            "divisions": "metropolitan",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false,
            "preferred_team_colors": true,
            "clock_rgb": "",
            "date_rgb": "",
            "flash_seconds": true
        },
        "weather": {
            "enabled": true,
            "view": "full",
            "units": "imperial",
            "duration": 30,
            "data_feed": "OWM",
            "owm_apikey": "584a45a2e53805b06ec4bb1ac18680ab",
            "update_freq": 5,
            "show_on_clock": true,
            "forecast_enabled": true,
            "forecast_days": 2,
            "forecast_update": 1
        },
        "wxalert": {
            "alert_feed": "NWS",
            "update_freq": 5,
            "show_alerts": true,
            "nws_show_expire": true,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 7,
            "show_on_clock": true
        }
    },
    "sbio": {
        "screensaver": {
            "enabled": false,
            "animations": true,
            "start": "22:00",
            "stop": "08:00",
            "data_updates": false,
            "motionsensor": false,
            "pin": 7,
            "delay": 30
        },
        "dimmer": {
            "enabled": true,
            "source": "software",
            "daytime": "6:00",
            "nighttime": "21:00",
            "offset": 0,
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "off_day",
            "sunset_brightness": 5,
            "sunrise_brightness": 60
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}

**Hardware you are using**
- Raspberry pi model  3b+ 
- $ uname -a
Linux hockey 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux

- 
- The Matrix driver (Adafruit Bonnet or Adafruit HAT)
- Bonnet 
- Anti-flicker mod or not? 
- Yes

**Software**
- Which version of the software are you on?
1,8,2

azaroth08, 2024-02-29:
Is this the image installation?

Overboard-code, 2024-02-29:
I downloaded the zip from : https://github.com/riffnshred/nhl-led-scoreboard/releases/tag/v1.8.2
unzipped it to /home/pi/nhl-led-scoreboard and ran scripts install.sh then nhl_setup

azaroth08, 2024-02-29:
Oh. Try removing all playoff settings from your config. The series ticker, and Stanley cup champions.  As well as the season and Christmas countdowns.

Overboard-code, 2024-02-29:
Changed it to:
"states": {
        "off_day": [
            "clock",
            "weather",
            "wxalert",
            "wxforecast",
            "standings",
            "scoreticker",
            "team_summary",
            "christmas"
        ],
        "scheduled": [
            "clock",
            "weather",
            "wxalert",
            "wxforecast",
            "scoreticker",
            "standings",
            "team_summary",
            "christmas"

And it works fine now.  Thanks a ton!

azaroth08, 2024-02-29:
Don't forget to remove the series ticker from post game and intermission

Overboard-code, 2024-02-29:
:-)