riffnshred/nhl-led-scoreboard #451: ModuleNotFoundError: No module named 'diskcache'

richmyers, 2024-03-07:
**Describe the bug**
Updated 1.7.x to latest version.  Testmatrix script works perfectly.  I see the goalpost animation....missing the Loading text....and then display is blank.

**Logs**
(https://paste.ubuntu.com/p/br5vjvX9tf/)
 utils import get_file
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 5, in <module>
    import diskcache as dc
ModuleNotFoundError: No module named 'diskcache'
Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 4, in <module>
    from data.scoreboard_config import ScoreboardConfig
  File "/home/pi/nhl-led-scoreboard/src/data/scoreboard_config.py", line 1, in <module>
    from utils import get_file
  File "/home/pi/nhl-led-scoreboard/src/utils.py", line 5, in <module>
    import diskcache as dc

**Your config file**
A screen shot of your config.json file can be really useful to help you. 
[image]
[image]
[image]

**Hardware you are using**
- Raspberry pi model -   Zero v1.1
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) - HAT
- Anti-flicker mod or not?  - yes

**Software**
- Which version of the software are you on?  1.8.2

falkyre, 2024-03-07:
Are you using the image?  If so, download the newest one. because the sb-upgrade script was not set up to handle the virtual environment that was introduced under V1.7.0 of the image.  If not, how did you do the upgrade?

pjockey, 2024-03-07:
Got the same error this morning after performing an update with git, then trying to run ./scripts/install.sh. Selected the upgrade option and the process stopped stating there was nothing to upgrade as it was already version 1.8.2. This is not the image, but from Riff's git page. 

I then performed `pip3 install diskcache` and the package installed. Trying to restart the board provided the same error.

I have Azroth's version also installed on the same card and it too has an error this morning. 
```
  File "/home/pi/nhl-led-scoreboard/src/data/scoreboard.py", line 152, in __init__
    home_skaters = len(overview["homeTeam"]["onIce"])
KeyError: 'onIce'
```
I'll have to switch to the MLB for the day!

azaroth08, 2024-03-07:
The onIce error is related to a change in the way the api formats it's data.  Riff is working a a fix for that.

As for the disk cache thing you are best off with starting with a brand new install of the newest image rather than trying to update from 1.7.x. The upgrade script doesn't account for the venv in the old version.

richmyers, 2024-03-07:
So I 

> Are you using the image? If so, download the newest one. because the sb-upgrade script was not set up to handle the virtual environment that was introduced under V1.7.0 of the image. If not, how did you do the upgrade?

So I downloaded the new image.  When I went to flash it on a 4GB SD card, BeleanaEtcher complained there was not enough space...330MB space needed.  I didn't have a larger card on hand so I installed the older image...from a couple days ago and updated using the sb-upgrade tool.  Will get a larger SD card and try it again.

falkyre, 2024-03-07:
Yah, bookworm lite seems to be bigger. In order to build the image, I needed 4gb of space. I haven't looked into making it smaller just yet.

richmyers, 2024-03-08:
Flashed the new image to a 32GB card and the only thing I did was connect to wifi.  Ran though setup...selected Team and enabled Supervisor.  This is the error I'm getting now.

d-scoreboard/src/data/scoreboard.py", line 152, in __init__
    home_skaters = len(overview["homeTeam"]["onIce"])
                       ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'onIce'
INFO: Validating config.json.....
INFO: config.json passes validation
Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 165, in <module>
    run()
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 160, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 80, in render
    self.__render_game_day()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 105, in __render_game_day
    self.scoreboard = Scoreboard(self.data.overview, self.data)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/nhl-led-scoreboard/src/data/scoreboard.py", line 152, in __init__
    home_skaters = len(overview["homeTeam"]["onIce"])
                       ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'onIce'

azaroth08, 2024-03-08:
> The onIce error is related to a change in the way the api formats it's data. Riff is working a a fix for that.
> 
This nhl is changing stuff on the fly. It will be fixed when possible.

richmyers, 2024-03-08:
No worries. Just wanted to report findings.  Thanks for all the work being done on this project!

richmyers, 2024-03-08:
Happy to report….it looks like the board is up and working.  Awesome work!

 

This email has been checked for viruses by Avast antivirus software.
www.avast.com

richmyers, 2024-03-11:
Have noticed the past few days....sometimes it works....sometimes it never progresses past loading.  NHL still making it difficult?