riffnshred/nhl-led-scoreboard #455: Stuck at hockey net with "LOADING"

harmgsn, 2024-04-18:
Team finished regular season - post season schedule not in API yet.  The "new day" after last game I start seeing:

list index out of range

in the stderr log file.  In the main log file:

[04/18/24 00:06:47] ERROR    list index out of range
[04/18/24 00:06:47] INFO     All preferred games are Final, showing the top
                             preferred game
[04/18/24 00:06:48] ERROR    list index out of range

This is the same with both live_mode set to true or false and debug set to true or false - relevant output does not change.

I am running a non-image install (ie: I installed raspbian then pulled from git) - master branch - latest update.

nick-danger, 2024-04-19:
It has to do with the API call to find the previous and next games. `nhl_api/info.py` at/about line 137 in `team_previous_game()`, there's a check for the truthiness of the results of fetching the team's schedule. It seems that if the team is done for the season, the API is returning an empty list. Two lines later, if the result was not truthy, the code nonetheless attempts to pull element 0 from the empty list and use it in an expression. I'm not certain what the original intent of that lump of code was, I've got a work-around that forces the previous game and next game data points to null when there's no data from the API call.

So if you're following a team that's out of the running, the team summary shows '--------' for the previous and next game data, but teams that still have upcoming games will show their next/previous games in the team summary. 

But the above change caused another issue. At/about line 152 of `data/scoreboard.py` in `Scoreboard.__init__()`, the code presumes that `overview["homeTeam"]` and `overview["awayTeam"]`contain `"onIce"` elements. I haven't figured out why they don't, but it's probably a side effect of returning null for previous and next games from above. I've got a quick work around for this, but it feels like a hack. I need to trace what happens to the results from `team_previous_game()` to see if I can deduce what a useful non-answer would look like.

riffnshred, 2024-04-19:
Fixed in 1.8.3

closed 2024-04-19 by riffnshred