riffnshred/nhl-led-scoreboard #457: OWM weather provider will stop working June 2024

falkyre, 2024-04-19:
Open Weather Map is deprecating their OneCall 2.5 API June 2024 and replacing it with a new API.  The core python library for the open weather map weather provider (pyowm) has not been updated to utilize the new API.  This issue is to track this and hopefully there will be an update to the pyowm library in the near future.

If you are using Environment Canada as a weather provider, this does not impact you.

xxxDrewedxxx, 2024-04-26:
Well this is not good!  I use One call on multiple devices...

xxxDrewedxxx, 2024-04-26:
Wait, I'm already on OneCall 3.0.  Everything works fine, if that helps.

falkyre, 2024-04-27:
I have found a new library but it will require some testing.  It is the library that the Home Assistant plugin has moved to and is using.